Naguissa/uTimerLib #2: Support SAMD

sabas1080, 2018-11-20:
Hi

Support SAMD21 or SAMD51??

Thanks!

Naguissa, 2018-11-20:
Hello,

I have none of that micros so I can try to code it but you should do the tests. Could you help me with that?

Thanks.

sabas1080, 2018-11-20:
of course i can help

Naguissa, 2018-11-21:
Thanks!

Any preference about timer number? I tend to use last timers. As currently I'm woking on SAMD21 and this chip has 3 and 5 TCs versions I was planning to use last one, TC2....

Cheers!

sabas1080, 2018-11-21:
TC2 or TC3 is good for me

Naguissa, 2018-11-23:
I've added support for ESP8266 (mostly tested) and SAMD21 on master.

Also I've updated some documentation and added SAMD21 datasheet to extras folder.

It's using Timer2 (3rd timer).

Could yo give it a try? At least it doen't complain when compiling....

sabas1080, 2018-11-23:
Compiling ok, but not working

Naguissa, 2018-11-28:
Fixed a copy-paste bug.

Could you try it?

You can try Serial examples, as in some ARCHs LED  doesn't work (at least on ESP8266, as it has no user LED, correctly defined at least).

If it fails again, can you describe the symptoms: nothing on Serial console, no repeated event, etc...

Naguissa, 2018-11-28:
Oh! You need to download master, as I've not created a release because it's not stable....

sabas1080, 2018-11-28:
Ok, testing

sabas1080, 2018-11-28:
I have tried with an MKR1300 and a Bast pro mini and not working examples of led and serial

Naguissa, 2018-11-29:
Could you try this example?

https://github.com/maxbader/arduino_tools/blob/master/libraries/timer_zero_tc_interrupt/timer_zero_tc_interrupt.ino

I've taken most of the code from there; maybe it's not working....

sabas1080, 2018-11-29:
not working blink and serial 🤔

sabas1080, 2018-11-29:
I have tried with this library too, but it does not work :(

https://github.com/adafruit/Adafruit_ZeroTimer

sabas1080, 2018-11-30:
Example https://github.com/maxbader/arduino_tools/blob/master/libraries/timer_zero_tc_interrupt/timer_zero_tc_interrupt.ino working in pin 11, in LED_BUILTIN not working.

But examples of uTimer not working in pin 11

Naguissa, 2018-12-03:
Hello again. I've deleted one extra flag, but code is equivalent to the one that works.

Check this:
 - Download / install master again
 - Open this example: uTimerLib_setInterval_s_example_led
 - Replace LED_BUILTIN for any known pin
 - Test it.

Should change pin status each 2 seconds...

Naguissa, 2018-12-04:
I found the problem. SAMD21 and SAMD51 follow different define structure to identify them. I'll finish some touches and I'll push some hanges including SAMD51 support.

Naguissa, 2018-12-04:
OK, SAMD21 and SAMD51 correctly added. Please, try it and give feedback.

Problem was I was using:

ARDUINO_ARCH_SAMD21
ARDUINO_ARCH_SAMD51

But architecture itself is always ARDUINO_ARCH_SAMD; I need to use:

_SAMD21_
__SAMD51__

(yes, it's not a typo, 21 has one surrounding "_" and 51 has two)

I've added a pragma telling support is experimental; I'll remove it if it's working and publish a new release.

Naguissa, 2018-12-28:
I received SAMD21 board yesterday. I've changed the code for a functional version.

But my perception is that timing is not perfect, it seems to be a little shorter than indicated (something like 1.8seconds if you specify 2 seconds).

Still, drafted a new release, 1.0.0 one.

To test it you may need to change LED_BUILTIN for a digital pin, if your board has no user LED (as my board happens to be).

sabas1080, 2018-12-28:
Gracias, ahora trabaja bien, he visto que tienes un foro en español, saludos

closed 2018-12-28 by sabas1080