Naguissa/uTimerLib #8: ARDUINO_STM32 - setInterval_s not correctly trigged

mignon-ne, 2019-10-21:
I have try with Maple compatible board.
When call with any value, it trig on every 1 second.

As example:
void setup() {
	Serial.begin(57600);
	TimerLib.setInterval_s(timed_function, 2);
	prevMillis = millis();
}

Will result in 1000

However, TimerLib.setInterval_us return correctly.
TimerLib.setInterval_us(timed_function, 2000000);
Will result in 2000

Naguissa, 2019-10-22:
OK, fixed.

There was a part missing when managing interrupts on STM32 for "seconds" functions.

Also, I enhanced examples with more correct coding.

closed 2019-10-22 by Naguissa

mignon-ne, 2019-10-22:
Hi Naguissa,

 

Wow, that is super fast.

Great to have your response.

It is working now.

 

 

One question, if you don’t mind.

Can I have 2 independent timer?

 

Best regards,

Kanok

Naguissa, 2019-10-22:
It's not possible with this library. But I suggest to do a broker function and call it at Maximum Common Divissor.

Timers is a scarce resource, so you shoukld preserve them. Also, Arduino functions and libraries are using some of them.

I try to use less common ones, so using even more timers will limit your options.

mignon-ne, 2019-10-22:
Thank you very much.

Not sure relate to an update or not, but seem the location is moved.
When compiled, it is completd and run successful.
However, I have warning. Possible my own issue,
=============
Resetting USB to switch back to runtime mode
Invalid library found in C:\Users\...\Documents\Arduino\libraries\uTimerLib: no headers files (.h) found in C:\Users\...\Documents\Arduino\libraries\uTimerLib
Invalid library found in C:\Users\...\Documents\Arduino\libraries\uTimerLib: no headers files (.h) found in C:\Users\...\Documents\Arduino\libraries\uTimerLib

Naguissa, 2019-10-22:
How did you installed/updated the library? You can do it via Library Manager on Arduino IDE.

You should check that location, seems to have any invalid copy there....

mignon-ne, 2019-10-22:
Solved, my own issue.

Found that your library name is "arduino_339753".
Probably from update, during open that folder "uTimerLib" and files.
Removed and re-install solved.
Thanks.