Naguissa/uTimerLib #9: STM32

AndKe, 2019-10-22:
I've used this library with STM32Duino library,   now I needed to switch to this, much better library:
https://github.com/stm32duino/Arduino_Core_STM32

- but uTimerLib is not compatible with it ?

```
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp: In member function 'int uTimerLib::setTimeout_us(void (*)(), long unsigned int)':
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:64:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp: In member function 'int uTimerLib::setTimeout_s(void (*)(), long unsigned int)':
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:92:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp: In member function 'void uTimerLib::_attachInterrupt_us(long unsigned int)':
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:179:18: error: 'TIMER_CH1' was not declared in this scope
   Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
                  ^~~~~~~~~
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:179:18: note: suggested alternative: 'TIMER_NUM'
   Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
                  ^~~~~~~~~
                  TIMER_NUM
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:179:29: error: 'TIMER_OUTPUTCOMPARE' was not declared in this scope
   Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
                             ^~~~~~~~~~~~~~~~~~~
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:179:29: note: suggested alternative: 'TIMER_OUTPUT_COMPARE'
   Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
                             ^~~~~~~~~~~~~~~~~~~
                             TIMER_OUTPUT_COMPARE
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:181:35: error: 'class HardwareTimer' has no member named 'setPeriod'; did you mean 'setOverflow'?
   uint16_t timerOverflow = Timer3.setPeriod(us);
                                   ^~~~~~~~~
                                   setOverflow
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:182:10: error: 'class HardwareTimer' has no member named 'setCompare'; did you mean 'setCount'?
   Timer3.setCompare(TIMER_CH1, timerOverflow);
          ^~~~~~~~~~
          setCount
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp: In member function 'void uTimerLib::_attachInterrupt_s(long unsigned int)':
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:463:18: error: 'TIMER_CH1' was not declared in this scope
   Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
                  ^~~~~~~~~
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:463:18: note: suggested alternative: 'TIMER_NUM'
   Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
                  ^~~~~~~~~
                  TIMER_NUM
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:463:29: error: 'TIMER_OUTPUTCOMPARE' was not declared in this scope
   Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
                             ^~~~~~~~~~~~~~~~~~~
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:463:29: note: suggested alternative: 'TIMER_OUTPUT_COMPARE'
   Timer3.setMode(TIMER_CH1, TIMER_OUTPUTCOMPARE);
                             ^~~~~~~~~~~~~~~~~~~
                             TIMER_OUTPUT_COMPARE
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:464:10: error: 'class HardwareTimer' has no member named 'setPeriod'; did you mean 'setOverflow'?
   Timer3.setPeriod((unsigned long int) 1000000);   // 1s, in microseconds
          ^~~~~~~~~
          setOverflow
/home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:465:10: error: 'class HardwareTimer' has no member named 'setCompare'; did you mean 'setCount'?
   Timer3.setCompare(TIMER_CH1, 1000000);
          ^~~~~~~~~~
          setCount
Multiple libraries were found for "uTimerLib.h"
 Used: /home/andre/Arduino/projects/libraries/uTimerLib
Multiple libraries were found for "GCS_MAVLink.h"
 Used: /home/andre/Arduino/projects/libraries/GCS_MAVLink
Using library uTimerLib at version 1.1.2 in folder: /home/andre/Arduino/projects/libraries/uTimerLib 
Using library GCS_MAVLink in folder: /home/andre/Arduino/projects/libraries/GCS_MAVLink (legacy)
exit status 1
Error compiling for board Generic STM32F1 series.
```

Naguissa, 2019-10-22:
I have no idea about that repository and why STM32duino is down...

I know STM32 / Arduino by STM32Duino.com (down) and Roger Clarckbourne, https://github.com/rogerclarkmelbourne/Arduino_STM32

I don't see him as contributor and your repository is not forked from Roger's one. In fact, "yours" is derived from ST development tools (Cube, etc) and "mine" is derived from Maple's Arduino-derived IDE.

I'll try your pointed repository and check if I can find its #defines in order to fix it....

If you want to try Roger's STM32 support, only to check, you can see the instructions I wrote here: https://translate.google.es/translate?sl=es&tl=no&u=https%3A%2F%2Fwww.foroelectro.net%2Felectronica-digital-microcontroladores-f8%2Fmicrocontroladores-basados-stm32-configuracion-arduino-ide-caracteristicas-t99.html

Cheers!

AndKe, 2019-10-22:
Yes, I've used Roger's STM32duino till recently.
Now I find that Arduino_Core_STM32 to be much better, more "complete" and compatible.

I do not know what happened to STM32duino
What I do know, is that the STM32duino forum closed for new sign-up - the "register" button and page was removed.  Then after an update, the option was back, but the forum owner never authorized new registrations.  A month after that again, the forum, webpage has been removed completely.  I guess he just got tired or something like that.

In any case, the Arduino_Core_STM32 is at least very much alive and being worked on.

Naguissa, 2019-10-22:
Oh, I had no idea....

I have a project where Maple Mini overclocked was getting too slow, I was thinking on start using asembler blocks.... So maybe with this new board support..... 

More arguments to try it ASAP.....

I'll check it, I think I'll need to check #define's and rename some functions; maybe check on old core files the constant definitions, but it shouldn't be too difficult.

Cheers!

Naguissa, 2019-10-23:
Added compatibility with both STM32 cores.

New core has been tested on Maple Mini (and compiled for few other boards).

New release published, 1.3.0

It should be available on the IDE's library manager in one hour approximately.

closed 2019-10-23 by Naguissa