Naguissa/uTimerLib #10: change TimerLib.setInterval_us(timed_function, myVar);

2kohm, 2019-11-29:
Hello,
is it possible to change 
`TimerLib.setInterval_us(timed_function, myVar);`
outside `void setup();` ?

i wan't to change the interval on the fly

Naguissa, 2019-11-29:
Hello,

Yes it's no problem. Calling that function will automatically ignore previous timed function so you don't need to call `clearInterval` before if you don't want to.

2kohm, 2019-11-29:
`#include "Arduino.h"
#include "uTimerLib.h"

volatile unsigned long int prevMillis = 0;
volatile unsigned long int actMilis = 0;
unsigned long ticker = 200000;
int ticks = 0;

	void
	timed_function()
{
	Serial.println(ticks);
}

void setup()
{
	Serial.begin(57600);
	//TimerLib.setTimeout_us(timed_function, ticker);
	prevMillis = millis();
}

void ticktimer(){
	TimerLib.setTimeout_us(timed_function, ticker);
	ticks++;

	if (ticks == 384){
		ticks = 0;
	}
}

void loop()
{
	ticktimer();
	actMilis = millis();
}
`
how can i count ticks in the timeframe defined in `ticker` ?

Naguissa, 2019-11-29:
I don't fully understand what are you trying to do, but anyway I've made it run.

First, few items:
 - It's adviced that any variable used in an interrupt (or any function called by an interrupt) has to be declared as volatile.

 - I deleted unused variables (actMilis and prevMilis).

 - To prevent calling timed function while it's still waiting i added a "running" flag, `inProgressFlag`

Revamped code looks like:

```
#include "Arduino.h"
#include "uTimerLib.h"

unsigned long ticker = 200000;

volatile int ticks = 0;
volatile bool inProgressFlag = false;

void timed_function() {
  Serial.println(ticks);
  inProgressFlag = false;
}

void setup() {
  Serial.begin(57600);
}

void ticktimer(){
  if (!inProgressFlag) {
    inProgressFlag = true;
    TimerLib.setTimeout_us(timed_function, ticker);
  }
  ticks++;
  if (ticks == 384){
    ticks = 0;
  }
}

void loop() {
  ticktimer();
}
```

The problem with previous code was that 	`TimerLib.setTimeout_us(timed_function, ticker);` was resetting the timer before `timed_function` was called.

2kohm, 2019-11-29:
Thank you,
i just want to make an counter that counts the ticker time interval

Naguissa, 2019-11-29:
Then If i'm not wrong it should be this way:

```
#include "Arduino.h"
#include "uTimerLib.h"

unsigned long ticker = 200000;

volatile int ticks = 0;
volatile bool inProgressFlag = false;

void timed_function() {
  Serial.println(ticks);
  inProgressFlag = false;
}

void setup() {
  Serial.begin(57600);
}

void ticktimer(){
  if (inProgressFlag) {
    ticks++;
  } else {
    inProgressFlag = true;
    ticks = 0;
    TimerLib.setTimeout_us(timed_function, ticker);
  }
}

void loop() {
  ticktimer();
}
```

2kohm, 2019-11-29:
thanks again for your fast replay.
let me make things clearer. I want to create an simple BPM Clock but with this old style micros i never get precise values.

```
unsigned long stepticks = 60000000 / (120 * 96);

	unsigned long timenow = micros();
	if (timenow - oldtime >= stepticks)
	{
		oldtime = timenow;
		tick++;
}
```
so, i want to get the timer and count for each micro senconds tick +1

2kohm, 2019-11-29:
ok, this works

volatile double cal1 = (120.00 / 60.0) * 96.0; // Use floating point to have decimals
volatile double cal2 = 2000000.0 / cal1;

void timed_function()
{
	Serial.print("\r ticks  ");
	Serial.print(ticks);
	inProgressFlag = false;
}

Naguissa, 2019-11-29:
Why don't you use setInterval and calculate us corresponding to desired BPM?

```
#include "Arduino.h"
#include "uTimerLib.h"

unsigned long us_per_desired_bpm = 123456;

void timed_function() {
  midi.send(0xF8);
  midi.update();
  ticks = 0;}

void setup() {
  Serial.begin(57600);
  TimerLib.setInterval_us(timed_function, us_per_desired_bpm);
}

void loop() {
  if (false) { // Here will be the condition to change BPM; us_per_desired_bpm should correspond to new BPM
    TimerLib.setInterval_us(timed_function, us_per_desired_bpm);
  }

}
```

us_per_desired_bpm = 1000000 * 60 / DESIRED_BPM

2kohm, 2019-11-29:
tanks again, this works great
```
volatile double cal1 = (121.00 / 60.0) * 24.0;
volatile double cal2 = 1000000.0 / cal1;
TimerLib.setInterval_us(timed_function, cal2);
```

Naguissa, 2019-11-29:
Perfect!

closed 2019-11-29 by Naguissa