Naguissa/uTimerLib #12: Error: no matching function for call to 'HardwareTimer::attachInterrupt

AndKe, 2020-07-06:
Hi, I am using STM32 cores for STM32F109  version 1.9.0   with uTimerLib 1.6.0
my code in in essence: 
#include "uTimerLib.h"
TimerLib.setInterval_us(timerEvent, 50000);

compilation fail with:

```

In file included from /home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:120:
/home/andre/Arduino/projects/libraries/uTimerLib/src/hardware/uTimerLib.STM32.cpp: In member function 'void uTimerLib::_attachInterrupt_us(long unsigned int)':
/home/andre/Arduino/projects/libraries/uTimerLib/src/hardware/uTimerLib.STM32.cpp:57:52: error: no matching function for call to 'HardwareTimer::attachInterrupt(int, void (&)(HardwareTimer*))'
   57 |     Timer3->attachInterrupt(1, uTimerLib::interrupt);
      |                                                    ^
        
......

In file included from /home/andre/Arduino/projects/libraries/uTimerLib/src/uTimerLib.cpp:120:
/home/andre/Arduino/projects/libraries/uTimerLib/src/hardware/uTimerLib.STM32.cpp: In member function 'void uTimerLib::_attachInterrupt_s(long unsigned int)':
/home/andre/Arduino/projects/libraries/uTimerLib/src/hardware/uTimerLib.STM32.cpp:98:52: error: no matching function for call to 'HardwareTimer::attachInterrupt(int, void (&)(HardwareTimer*))'
   98 |     Timer3->attachInterrupt(1, uTimerLib::interrupt);
      |                                                    ^
```

Any ideas why ?

Naguissa, 2020-07-06:
STM32F109? I don't find its datasheet, any info or that model on Arduino boards selector. Maybe it's 103?

AndKe, 2020-07-06:
Yes, my mistake  STM32F103 is correct

AndKe, 2020-07-06:
Also, FWIW, I had this library/code to work fine half a year ago or so (older versions of STM32 core , this library, and maybe Arduino)

Naguissa, 2020-07-06:
> 
> 
> Additional information: this library/code worked fine half a year ago or so (older versions of STM32 core , this library, and maybe Arduino)

Yes, it worked: #9 

Seems they have changed timers. I'll fix it asap.

Naguissa, 2020-07-08:
I've just released version 1.6.1 trying to fix that issue.

There're some changes on compiling process and timer usages.

But I'm having trouble testing it, on Roger clark and ST's core, also on blue pill and Maple mini....

So please, comfirm me if all issues are solved....

Cheers!

AndKe, 2020-07-10:
I tried 1.6.1 on BluePill with F103C8T6 

This, modified example:
```
/**
 * uTimerLib example
 *
 * @author Naguissa
 * @url https://www.github.com/Naguissa/uTimerLib
 * @url https://www.foroelectro.net
 */

#include "Arduino.h"
#include "uTimerLib.h"

#ifndef LED_BUILTIN
	// change to fit your needs
	#define LED_BUILTIN PC13
#endif

volatile bool status = 0;

void timed_function() {
	status = !status;
}

void setup() {

  pinMode(LED_BUILTIN, OUTPUT);
  int x=0;
  while (x<5) {
  digitalWrite(LED_BUILTIN, status);
  delay(100);
  status = !status;
  x++;
  }
 
	TimerLib.setInterval_s(timed_function, 2);
}

void loop() {
	digitalWrite(LED_BUILTIN, status);
}
```

first I see rapid flashing as expected. (proving PC13 is the right pin)
- then the LED stays off.
(the 2s function is not called)

Naguissa, 2020-07-11:
Fixed and checked on 1.6.2 release.

If you have any problem check LED pin on examples.

AndKe, 2020-07-12:
Ok, I tried the stock 1.6.2 's uTimerlib_setinterval_s_example_led, only modifying led to PC13  - did not work.
So I tried the "serial" variation of the example - did not work...
So I added one println to it like:
```
void setup() {
	Serial.begin(57600);
  Serial.println("eiudhuehhd !");
	TimerLib.setInterval_s(timed_function, 2);
```

And after each reset -  my terminal sees "eiudh"   then no more characters.
To me it seems like the setInterval_s call crashes the STM somehow so that the string is never completed.

AndKe, 2020-07-17:
@Naguissa please look into this - it still won't work with bluepill.

Naguissa, 2020-07-18:
Finally I've sorted my environment. I had to reinstall STCore because something was (probably) corrupt, but also install stm32cubeprogrammer crap in order to use STLink....

Results:

 - Maple Mini - 1st one tested, LED OK, serial OK (after reinstalling STCore).
 - Blue Pill - Tested after installing crappy stm32cubeprogrammer. LED OK, Serial OK (this one tested when I was writing this text, as I thought I had the faulty resistor blue pill, but it works...)

At first none worked, but I don't know if it was corrupted STCore or something wrong with my microcontrollers. Once I reinstalled and Maple mini worked once, every test has been passed without problems.

closed 2020-07-18 by Naguissa

AndKe, 2020-07-19:
@Naguissa 
Sorry for botherting you so much while trying to pinpoint the issue here, I do not know what is it's cause, but I suspect many more can have the same issue - as is it not clearly visible in projects that use, but not fully depend on the timer.

I closed Arduino.
removed: 
/tmp/arduino*
/home/andre/Arduino/projects/libraries/uTimerLib
/home/andre/.arduino15/packages/STM32

- reinstalled uTimerlib 1.6.2 and STM32Cores 1.9.0

My compilation completesm but is not working - compiler output:
https://gist.github.com/AndKe/ed9dd5ca9f9e9f0561a1e21c2a89197b

Arduino_Core_STM32
  Board: Generic STMF1
  Board Part : Bluepill F103C8
  U(S)ART support: Enabled(generic Serial)
  USB Support: CDC(generic "serial" supersede U(S)ART"
  USB Speed: Low/Full Speed
  Optimize: Smallest (-Os default)
  C Runtime Library : newlib Nano(default)
  Upload Method: STM23CubeProgrammer(SWD)  (ST_LINKV2)

My build settings:
```
  Arduino_Core_STM32
  Board: Generic STMF1
  Board Part : Bluepill F103C8
  U(S)ART support: Enabled(generic Serial)
  USB Support: CDC(generic "serial" supersede U(S)ART"
  USB Speed: Low/Full Speed
  Optimize: Smallest (-Os default)
  C Runtime Library : newlib Nano(default)
  Upload Method: STM23CubeProgrammer(SWD)  (ST_LINKV2)
```

Can you please provide me your compiler output, or think of what else could be the difference here ?
- I do not have STM32duino installed, maybe you have it?

Naguissa, 2020-07-21:
https://gist.github.com/Naguissa/486ca6395cdf1ee40f111b487e703f15

I have STM32duino, but it doesn't interfere with STCore

Naguissa, 2020-07-24:
I've found few uncommited changes. Commited and created new release: https://github.com/Naguissa/uTimerLib/releases/tag/1.6.3

AndKe, 2020-07-25:
I were midway in a desperate full backup/clean install to try to figure out how it could work for you, not me.
Glad to see there was a perfectly logical explanation, 1.6.3 works again perfectly.

Thank you.