Naguissa/uTimerLib #13: ESP32 Timer precision

raffaeler, 2021-04-22:
During compilation I slapped my face into this message
```
"ESP8266 / ESP32 can only reach a ms resolution so any us interrupt will be rounded to that"
```
Reference: https://github.com/Naguissa/uTimerLib/blob/848ebf133cdd1bcef20303276b59b6659cdc07a6/src/uTimerLib.h#L108

Apparently, ESP32 does not have this limitation which sounds very weird for any microcontroller.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html#obtaining-current-time

Maybe the one used in this library is not the the right timer.
Am I missing something?

Thank you!

Naguissa, 2021-04-22:
Hello,

As stated on header, I use RTOS Timer to provide support: https://github.com/Naguissa/uTimerLib/blob/848ebf133cdd1bcef20303276b59b6659cdc07a6/src/uTimerLib.h#L13

This timer has 1ms resolution, hence that message.

I've looked your documentation and it's possible to use a hardware timer in a secure way, so I'll modify the code to do so. Thanks for the hint!

Cheers!

P.S.: ESP Arduino support is based on FreeRTOS, that's why I didn't want to touch hardware directly: https://techtutorialsx.com/2017/05/06/esp32-arduino-using-freertos-functions/

raffaeler, 2021-04-22:
That's great, thanks for the prompt answer.
Given the ESP32 has a supersonic speed in comparison to the standard microcontroller, I bet the ability of controlling more fine-grained timer is popular :)

Naguissa, 2021-06-25:
Hello again!

I've integrated OS hardware timer on ESP32. ESP8266 lacks that class, so I've splitted both and ESP8266 still uses ticker.

I'll check if ESP8266 has any similar class when I have some time....

Thanks for the hint!

Cheers!
Daniel.

closed 2021-06-25 by Naguissa