Naguissa/uTimerLib #14: clearTimer() does not work in 1.6.5 ( it does in 1.6.4 )

m-menon, 2021-07-02:
hi,

`TimerLib.clearTimer();`

doesn't work in 1.6.5. I have tried it with this simple example. The same thing works well with version 1.6.4 - tried with both setInterval and setTimeout.

**Example 1 (setInterval)**
```
#include "Arduino.h"
#include "uTimerLib.h"

volatile unsigned long int prevMillis = 0;
volatile unsigned long int actMillis = 0;

void timed_function() {
  Serial.println(actMillis - prevMillis);
  prevMillis = actMillis;
  TimerLib.clearTimer();
}

void setup() {
  Serial.begin(115200);
  Serial.println("Timer started!");
  TimerLib.setInterval_s(timed_function, 2);
  prevMillis = millis();
}

void loop() {
  actMillis = millis();
}
```

**Example 2 (setTimeout)**
```
#include "Arduino.h"
#include "uTimerLib.h"

volatile unsigned long int prevMillis = 0;
volatile unsigned long int actMillis = 0;

void timed_function() {
  Serial.println(actMillis - prevMillis);
  TimerLib.clearTimer();
}

void setup() {
  Serial.begin(115200);
  Serial.println("Timer started!");
  TimerLib.setTimeout_s(timed_function, 2);
  prevMillis = millis();
}

void loop() {
  actMillis = millis();
}
```

Naguissa, 2021-07-02:
Fixed!

m-menon, 2021-07-02:
Wow, fantastic! Thank you.

closed 2021-07-02 by m-menon

Naguissa, 2021-07-02:
I just was on the computer and that notification arrived; it was quite easy... ;-)

m-menon, 2021-07-02:
> I just was on the computer and that notification arrived; it was quite easy... ;-)

Right place at the right time! ... :-D Cool, thanks once again.