Naguissa/uTimerLib #15: Timer crash in esp32

raffaeler, 2021-09-14:
I just downgraded a project from PlatformIO to Arduino IDE.
- on PlatformIO I could install the 1.6.6
- on Arduino IDE I now can see the 1.6.5 only

This project starts and stop a timer (50ms interval) with two serial port commands, very easy.
- Using the 1.6.6 the second time I start the timer, the ESP32 crashes.
- Using the 1.6.5 it works nicely

Note: I could understand the cause was the timer because I decoded the ESP32 backtrace using a tool.

I can't understand from the diff, how this can cause the crash, but apparently it does.
https://github.com/Naguissa/uTimerLib/compare/1.6.5...1.6.6#diff-e2a251b850a8484de4afd48e6d37cb2b3600da975fcfca21e8d35935c60c0b17R92

Naguissa, 2021-09-15:
I'll check ASAP, seems a problem with cleartimer / create a new one

Naguissa, 2021-09-15:
Solved on new 1.6.7 release: https://github.com/Naguissa/uTimerLib/releases/tag/1.6.7

It will appear on Library Manager approximately in 2-3 hours.

closed 2021-09-15 by Naguissa