Naguissa/uTimerLib #16: STM32 compilation fails with error about Timer3

joncutting, 2022-01-04:
uTimerLib.h line 150 references `_VARIANT_ARDUINO_STM32_` and this is for some reason not defined when I use Nucleo32 / NucleoG031K8, so the library compile borks because it doesn't know anything about Timer3. 

As a workaround I created a platform.local.txt file with a single line

`compiler.extra_flags=-mcpu={build.mcu} {build.flags.fp} -DUSE_FULL_LL_DRIVER -D_VARIANT_ARDUINO_STM32_ -mthumb "@{build.opt.path}"`

To clean this up I would suggest to use `ARDUINO_ARCH_STM32` instead of `_VARIANT_ARDUINO_STM32_`.

Naguissa, 2022-01-04:
Wich core are you using? Roger Clark or ST core?

Thanks!

joncutting, 2022-01-04:
It’s the ST core 2.2

Naguissa, 2022-01-12:
Solved on new 1.7.0 release, https://github.com/Naguissa/uTimerLib/releases/tag/1.7.0

Cheers!

closed 2022-01-12 by Naguissa