Naguissa/uTimerLib #23: STM32F103 freezing after a short while - worked fine before.

AndKe, 2025-03-13:
I set a timer for 50ms 
The failed function does not need to do anything - it usually just set a variable.

After a short while, the uC will freeze.

This worked fine half a year ago, when I last compiled this program. 
now it fails after a while.

(I omitted the use of this library to solve it.)

Naguissa, 2025-03-13:
May marking the variable used inside interrupt as "volatile" solve the issue?

If you share a failing code would be useful; still I'll try to recreate.

AndKe, 2025-03-13:
I also tried replacing the variable with just delayMicroseconds(1) 

The program is big with some custom libraries and dependencies, but the gist is:
stm32_core is used.
MAVLink library is used (not uC specific)
two HW serial ports are used, one in one-wire bidirectional mode.
CDC serial is used.
a few other libraries that I could disable also proved to be invariant to this issue.

All I needed/used was:

TimerLib.setInterval_us(timer20hzEvent, 50000); 

void timer20hzEvent() {
   timer20hz = true;
}

This had worked for years. The main loop checked variable timer20hz.
As mentioned, I also tried to replace setting the variable with just a small delay.
The main loop runs for a short while (I did not time it - but I doubt it was as much as one sec)  - then everything stopped.

It is years since I changed the essential parts of this code.

This could very well be some super-odd case of some incompatibility between updated libraries/compiler...

Naguissa, 2025-03-13:
If you check the code, it has been 3 years since last change: https://github.com/Naguissa/uTimerLib/tree/master/src/hardware

Using delay inside interrupts is a no-go.

timer20hz variable should be declared as volatile.

As stmt2_core I understand official ST core, right? It's not Roger Clark one.

As implementation has not changed I suspect something changed on ST core; I'll try to recreate when I have some spare time.

AndKe, 2025-03-13:
Yes,  official ST core I also tried older Arduino IDE 2.3.2, and older ST-code, and failed to make it work.
ST Core had more releases recently, but if up/downgrade system works as intended - that's not it.
Thank you.

Naguissa, 2025-04-26:
No response, I understand it's fixed by my tests.

closed 2025-04-26 by Naguissa