Naguissa/uTimerLib #24: Not working on Arduino Pro Mini 3.3 V, 8 MHz

benjaminemhf, 2025-04-13:
Sadly the library doesn't work on the Arduino Pro Mini 3.3 V, 8 MHz (ATmega 328P) variant.

It works fine on an Arduino Uno 16 MHz, and it works on the Arduino Pro Mini 8 MHz as well if I set the the processor version to 5 V, 16 MHz, but of course then the timings are not correct... Tried it on multiple boards.

I'm guessing something might be wrong with prescaler selection or 'us' calculation from the 'f_cpu' variable but I'm not sure.

Kind regards,
Benjamin

Naguissa, 2025-04-14:
It's quite strange. f_cpu is used to adapt timers to current CPU speed, so using it as 16MHz should drive it correctly (if your microcontroller supports it).

Could you add your relevant code? I think it may be because too short delays or problems when using the interrupts, as using serial o delay in them.

Have you tried uTimerLib_setInterval_us_example_led example?

benjaminemhf, 2025-04-14:
Hi, thank you for the response. Yes, I've tried all of the examples.

With the _uTimerLib_setInterval_us_example_led_ the LED turns ON and stays ON.
With the _uTimerLib_setTimeout_us_example_led_ the LED turns ON for a very brief moment and then stays OFF.
With the _uTimerLib_setTimeout_us_example_serial_ and _uTimerLib_setTimeout_s_example_serial_ the time reported on the serial monitor is 0.

My use case is very simple and similair to the setTimeout_us example, I just need an LED to light up for a short period of time (about 30 ms) without blocking code with delays. The ISR only turns OFF the LED.

benjaminemhf, 2025-04-14:
All of this works fine on the Arduino Uno, and it has exactly the same microcontroller as the Pro Mini (ATmega328P), the only difference is the external crystal oscillator: 16 MHz on the Uno vs 8 MHz on the Pro Mini.

I've disconnected all external components from the Arduinos and tried different pins for the LED. Really can't figure this out.

Naguissa, 2025-04-15:
I suspect it may be because frequency division. I've draft a new release trying to fix it.

closed 2025-04-17 by Naguissa